R0rt1z2/lkpatcher #1: Is there any workaround for SBC devices?

Itz-Broken-xd, 2023-10-04:
Device : Oppo f15
OS : ColorOs 11.1 (based on a11)

Lk image : [lk.bin](https://xfl.jp/YbWmei)
Expbd : [expdb.bin](https://xfl.jp/RUiduM)

Note : the above lk image is booting(stock one) 
But it still shows image auth fail in expdb. Any fix?

Avb is patched by magisk. 

Btw my device has two bootloader partitions `lk` and `lk2`.

R0rt1z2, 2023-10-06:
Can you please send the tool's output?

Itz-Broken-xd, 2023-10-12:
I used online patcher and it worked fine(it showed no error). But when flashed it wont boot the device

closed 2023-10-23 by R0rt1z2

CE1CECL, 2024-10-16:
@n0t-Human Just want to make sure (sorry for old bump) you did this in mtkclient:
```
python mtk.py da seccfg unlock
```
This issue only happens to me when its locked on TCL 5102L (A30)

n0t-Human, 2024-10-16:
@CE1CECL 

I tried unlocking using mtkclient but it showed some error. So I unlocked it using Meadiatek Easy Tool. 

Ss:
[image]

n0t-Human, 2024-10-17:
> Don't advertise shitty MTKClient rebrands, please.

Sorry, I didn't mean to. I provided the details if he could find any workaround. I'm not sponsored by those rebrands.🥲