R0rt1z2/lkpatcher #3: Request for Documentation on JSON Schema of Patches

animegamer4422, 2023-10-22:
Hello,

I am working with LKPatcher to patch MTK LK images in order to resolve issues with dm-verity, fastboot access, and the red state error. This tool has been really helpful, but I'm currently facing a challenge with understanding how the patches work due to lack of documentation on the JSON schema used for patches.

For my particular use case, I am looking to create a patched LK image that eliminates the orange state error by locking the bootloader upon flash. This will allow me to resolve the errors while also being able to unlock the bootloader using the official method later on.

I believe having documentation regarding the JSON schema of patches would greatly assist in understanding how to create custom patches to meet specific needs. This documentation would serve not only me but also other developers who may want to leverage this tool for similar or other custom solutions.

Would it be possible to provide documentation or examples on:
1. The structure and properties of the JSON schema used for patches.
2. How the patches interact with the LK images.
3. Steps or guidelines on creating custom patches, especially for locking and unlocking the bootloader.

I appreciate the work done on this tool and hope that additional documentation can make it more accessible and easy to use for the community.

Thank you for your time and consideration.

Best regards

luoyuancheng, 2023-10-22:
Hey,guys.This tool is used with annother python lib that created by this author,you can find that from his repositories.The name is "liblk".In the meantime,the  author also privide a online tool with the same function but used easily, it is introduced  in the readme file.

animegamer4422, 2023-10-22:
I used the online tool and it works great
 I just wanted to know how did he knew the location of the patches

Like how to figure that out is what I want to know

AntiEngineer, 2023-10-23:
https://github.com/R0rt1z2/lkpatcher/assets/16330761/d0298591-e69c-4ebd-bab0-832a32093f65

R0rt1z2, 2023-10-23:
> Hello,
>
> I am working with LKPatcher to patch MTK LK images in order to resolve issues with dm-verity, fastboot access, and the red state error. This tool has been really helpful, but I'm currently [...]

The schema provides a structured catalog for all the patches. The main categories, such as `fastboot `and `red_state`, are just identifiers to be able to easily recognize all the patches. Within each category, there are HEX sequences which represent ARM instructions. These sequences might vary based on the Android version, hence multiple entries can be found under a single category. Each of these correspond to a replacement, typically in the same format. For the provided entries, the default replacement is `00207047`, which stands for the ARM instructions `mov r0, #0x0; bx lr` (`return 0`).

The patches are directly applied to the images. This alters the actual contents of the LK, which effectively modifies the instruction flow or the underlying LK code, unlike certain other exploits which might work as some sort of _overlay_, adding a layer on top without changing the original code. Patching compromises the image's integrity and, as a result, if your device uses SBC, it will reject the patched image.

There isn't really a one-size-fits-all guide for crafting custom patches. What I personally did was to open the LK image with Ghidra and look for the function that decides whether fastboot mode is available or not. Once you spot the signature (the first 4-8 bytes) of the function you're aiming to patch, just throw a new category into the JSON schema and slap in your replacement.

DeadPool-4422, 2023-10-23:
I can see the data in a Hex editor
 Can I convert the Hex data to assembly code and then understand the stuff to make changes?

DeadPool-4422, 2023-10-23:
Thanks for explaining the process and tools you used

Have a great day @R0rt1z2

R0rt1z2, 2023-10-23:
https://github.com/R0rt1z2/lkpatcher/commit/ab9f8bb5b17aebf35bcfdf83db084a15cc50a9cf