R0rt1z2/lkpatcher #4: dm-verity message not disabled.

hereblur, 2023-10-22:
Hi, thanks for great work.

On my device: Realme C25 / RUI 4.0 / Android 13
Unlocked BL with mtkclient.

After apply the patch, fastboot working. then orange state is disabled. but the dm-verity message still showing and required power button push to continue.

> [root] [INFO] [2023-10-23 02:19:22] MediaTek bootloader (LK) patcher - version: 3.0.0 by R0rt1z2.
[lkpatcher.patcher] [INFO] [2023-10-23 02:19:22] Successfully loaded 5 patches.
[lkpatcher.patcher] [WARNING] [2023-10-23 02:19:22] Skipping invalid needle 2de9f04fadf5ac5d.
[lkpatcher.patcher] [INFO] [2023-10-23 02:19:22] Successfully applied 4 patches.
[root] [INFO] [2023-10-23 02:19:22] Patched image saved to lk-patched.bin.

R0rt1z2, 2023-10-22:
Send me your LK image.

hereblur, 2023-10-23:
Here  is [lk.bin](https://paste.c-net.org/WedgedSphere)

May I know how you read or decompile it? 
I ask because I want to debug some problem like boot loop on non-stock images.  not sure if it `lk` related though. 
Thanks,

R0rt1z2, 2023-10-23:
Give it another shot using https://github.com/R0rt1z2/lkpatcher/commit/cb9c72d8f075d7bfc69f92e2cdfff3f0d02e6d16. I've also put together a short guide for the decompilation process in https://github.com/R0rt1z2/lkpatcher/commit/243bd8d294acec2b1b9ed60ddce1c4d7ede2378d.

hereblur, 2023-10-24:
It's working now, thank you.

Also thanks for the guide. it's really useful.

closed 2023-10-24 by hereblur