R0rt1z2/lkpatcher #5: Can't enter fastboot, stuck in the entry interface

Chengrose, 2023-10-24:
Hello,
I want use this tool to solve my issues with fastboot access,but when I try entring fastboot mode,my device stuck in the entry interface,just like this[https://github.com/Chengrose/lk/blob/main/QQ%E5%9B%BE%E7%89%8720231024091550.jpg](url).
this is my lk.img[https://github.com/Chengrose/lk/blob/main/lk.img](url).
Thank you for your time and consideration.
Best regards

R0rt1z2, 2023-10-24:
That's completely normal, and that message indicates that your device is already unlocked (or that you have already patched LK). Make sure you have fastboot drivers installed and check for unknown devices in the device manager.

Chengrose, 2023-10-24:
> That's completely normal, and that message indicates that your device is already unlocked (or that you have already patched LK). Make sure you have fastboot drivers installed and check for unknown devices in the device manager.这是完全正常的，该消息表明您的设备已解锁（或者您已经修补了LK）。确保您已安装快速启动驱动程序，并在设备管理器中检查未知设备。
So it was, I forgot to install the fastboot driver and there is no gui with the fastboot mode of my device.Thank you for your help!

closed 2023-10-24 by R0rt1z2