R0rt1z2/lkpatcher #7: is it possible to chainload a boot.img file with different linux kernel or android kernel through your tool?

p2pdownsbrz, 2024-03-01:
Hey @R0rt1z2 , is it possible to chainload a boot.img file with different linux kernel or android kernel through your tool? I didn't really understand why patch the LK partition, I didn't understand what changes on the device. I would be grateful if you explain it to me. I came across your repository while looking for a way to boot an arm64 linux distro, without having to compile a kernel as recommended by halium.org...
my device is mt6785 (infinix-x695c with android 11 gki-version1) I was unable to compile the kernel specifically for my device to use halium, I found a halium-11 gki but the boot partition of my smartphone is small, even though it has a vendor-partition boot... so I thought I could flash some GKI kernel, but gki kernel can't be flashed... the boot partition is 32MB~ I'm looking for alternative ways... not necessarily easier. Sometimes what works for one person may not work for another person.

Chengrose, 2024-03-01:
Hey,bro!This tool is used to let the fastboot mode that hided by oems  appears,mainly for some mtk devices come from oppo,realme.As for you idea about boot linux  on arm,it maybe help you to learn about droidian that it is a debian distribution for mobile phone.You can compile you kernel to boot the gsi image provided by that project.I hope that my advice is useful.

p2pdownsbrz, 2024-03-01:
@Chengrose, Thanks, I'll take a look.

closed 2024-03-01 by p2pdownsbrz

R0rt1z2, 2024-03-05:
> Hey @R0rt1z2 , is it possible to chainload a boot.img file with different linux kernel or android kernel through your tool? I didn't really understand why patch the LK partition, I didn't understand what changes on the device. I would be grateful if you explain it to me. I came across your repository while looking for a way to boot an arm64 linux distro, without having to compile a kernel as recommended by halium.org... my device is mt6785 (infinix-x695c with android 11 gki-version1) I was unable to compile the kernel specifically for my device to use halium, I found a halium-11 gki but the boot partition of my smartphone is small, even though it has a vendor-boot partition... so I thought I could flash some GKI kernel, but gki kernel can't be flashed... the boot partition is 32MB~ I'm looking for alternative ways... not necessarily easier. Sometimes what works for one person may not work for another person.

To learn more about how this tool works, check out my blog post [here](https://blog.r0rt1z2.com/patch-mediatek-bootloader-images-lk.html). Also, if you're interested in chainloading, it's definitely doable. You might want to consider using a custom LK payload, though. For details and code examples, take a look at my [kaeru project](https://github.com/R0rt1z2/kaeru) on GitHub.