R0rt1z2/lkpatcher #10: does your patch repair dm-verity corruption ?

Foufure13, 2024-03-13:
Hello i need some help for my device bv4800 mt6761v

and I'd like to stop having to press the power button a second time to boot. 

because of the error: 
dm-verity corruption
Your device is corrupt.
It can’t be trusted and may not work
Press power button to continue. Or, device will power off in 5s

so I wanted to know if your tool could handle it

[lk_a.zip](https://github.com/R0rt1z2/lkpatcher/files/14590863/lk_a.zip)
[boot_a.zip](https://github.com/R0rt1z2/lkpatcher/files/14590866/boot_a.zip)

R0rt1z2, 2024-04-28:
It should, yes. Just try the tool.

Foufure13, 2024-05-10:
I didn't succeed with the tool 
but by following the steps with your tuto and ghidra I was able to solve my problem.

closed 2024-05-10 by Foufure13

NCLnclNCL, 2025-05-22:
> I didn't succeed with the tool but by following the steps with your tuto and ghidra I was able to solve my problem.

how to bro