R0rt1z2/lkpatcher #11: There is a BL Lock problem about HarmonyOS.

Ceyase, 2024-04-04:
I have unlocked seccfg, but I did not successfully unlock Bootloader. I'm sure this is because Huawei modified lk, maybe not. I tried using Ghidra to analyze this file, but I didn't get any results. I need to use Magisk, so I just need to skip the verification - if possible.
Before receiving your response, I will do my best to decode this lk.bin. By the way, your blog and online tool website are both damaged. 

Some information may be useful:
[lk.zip](https://github.com/R0rt1z2/lkpatcher/files/14866254/lk.zip)
[image]

```shell
[Ceyase@CeyaseArch ~]$ adb reboot bootloader
[Ceyase@CeyaseArch ~]$ fastboot oem unlock
(bootloader) cmd is invalid
FAILED (remote: 'CMD_OEM FAILED!!')
fastboot: error: Command failed
[Ceyase@CeyaseArch ~]$ fastboot boot /home/Ceyase/Desktop/mtkclient/honor_backup/recovery.bin 
Sending 'boot.img' (46080 KB)                      OKAY [  1.141s]
Booting                                            FAILED (remote: 'Command not allowed')
fastboot: error: Command failed
[Ceyase@CeyaseArch ~]$ fastboot getvar all
getvar:all                                         FAILED (remote: 'Command not allowed')
Finished. Total time: 0.000s
[Ceyase@CeyaseArch ~]$ 

```

Ceyase, 2024-04-04:
bro, you don't need to solve this problem anymore. Because this phone is already broken - completely. I accidentally dropped this phone from upstairs, so ...now the screen is completely useless

R0rt1z2, 2024-04-28:
lol

closed 2024-04-28 by R0rt1z2