R0rt1z2/lkpatcher #14: patch for mt6893,can this tool remove signature verification?

dysf888, 2024-05-08:
I tried to modify the logo to remove the logo of the unlocked device, but when I flashed the modified Logi.bin, the device would not turn on and the screen would remain black.I guess LK has signature verification on the logo during the startup phase, but I don't know how to modify files.

LK.img get from official MIUI package
LK(a/b).bin  get from official MIUI mtkclient
device RedMi Note 10 Pro  MTK 1200 (mt6893)

[LK.zip](https://github.com/R0rt1z2/lkpatcher/files/15247766/LK.zip)

R0rt1z2, 2024-05-08:
I believe LK is also verified. Have you tried altering at least one bit and then reflashing the modified LK onto the device to confirm if it can boot with the patched LKs?

dysf888, 2024-05-08:
> I believe LK is also verified. Have you tried altering at least one bit and then reflashing the modified LK onto the device to confirm if it can boot with the patched LKs?

```
C:\Users\dysf8>certificatehashfileD:\MIUI\ROMs\Note10-PRO_256-MTK\lka.bin
SHA1'sD:\MIUI\ROMs\Note10-PRO_256-MTK\lka.binhash:
32fa3ad89f902921b8c5727ba31b01b1bf12853e
TheCertUtil:-hashfilecommandcompletedsuccessfully.

C:\Users\dysf8>certificatehashfileC:\Users\dysf8\Desktop\LK\lka.bin
SHA1'sC:\Users\dysf8\Desktop\LK\lka.binhash:
2d8e065b2a9abfb384cb2eb13794972420a5bc76
TheCertUtil:-hashfilecommandcompletedsuccessfully.

C:\Users\dysf8>certificatehashfileC:\Users\dysf8\Desktop\LK\ReadAfterFlash\lka.bin
SHA1'sC:\Users\dysf8\Desktop\LK\ReadAfterFlash\lka.binhash:
2d8e065b2a9abfb384cb2eb13794972420a5bc76
TheCertUtil:-hashfilecommandcompletedsuccessfully.
```

Thank u for your reply
I added a lot of A with Winhex and it can boot up normally
[image]

R0rt1z2, 2024-05-09:
Try [this LK image](https://github.com/R0rt1z2/lkpatcher/files/15268282/lk.zip). I basically forced `img_auth_stor` to return 0 (success) no matter what. For future reference, you can use the following JSON:
```json
{
    "fastboot": {
        "2de9f04fadf5ac5d": "00207047",
        "f0b5adf5925d": "00207047"
    },
    "dm_verity": {
        "30b583b002ab0022": "00207047"
    },
    "orange_state": {
        "08b50a4b7b441b681b68022b": "00207047"
    },
    "red_state": {
        "f0b5002489b0": "00207047"
    },
    "image_verify": {
        "2de9f0470f469046": "00207047"
    }
}
```

dysf888, 2024-05-10:
Thank you for your help!

closed 2024-05-10 by dysf888

R0rt1z2, 2024-05-11:
Did it work?

dysf888, 2024-05-11:
> Did it work?

Hey, I'm really sorry but I didn't get to try it out. I accidentally spilled some Coke on it. Anyway, thanks again for helping me.

R0rt1z2, 2024-05-11:
💀