R0rt1z2/lkpatcher #15: Cannot reboot to fastbootd mode

fayaz-modz, 2024-05-25:
model: mt6768

I'm stuck in bootloader mode, (i am able to goto recovery but cannot go to fastbootd mode) when i type fastboot reboot fastboot, it reboots to bootloader only

R0rt1z2, 2024-05-25:
I'm sorry, but what does this have to do with my tool? Have you patched your LK with it? I need more details, such as the phone model, the brand, and the Android version.

fayaz-modz, 2024-05-26:
Yes, I have patched my lk_a file. I'm able to boot to fastboot mode. When I type fastboot devices, it says 'fastboot' not bootloader. And I'm trying to boot to fastbootd but I come back to the same mode.

My device is realme c55.

R0rt1z2, 2024-10-28:
I believe this is unrelated to lkpatcher so I'm just gonna close this issue.

closed 2024-10-28 by R0rt1z2