R0rt1z2/lkpatcher #20: requesting device specific patch for (mrd-lx2)

r4db, 2024-06-29:
[lk.zip](https://github.com/user-attachments/files/16039171/lk.zip)
how to get image.txt?

fyhbauyg, 2025-05-10:
> [lk.zip](https://github.com/user-attachments/files/16039171/lk.zip)您好，您能否为 MRD-LX2 华为 Y6 Pro 2019 修补我的 LK

lkpatcher is oppo/realme tool,i not can find base address patch this lk.img

R0rt1z2, 2025-05-10:
> > [lk.zip](https://github.com/user-attachments/files/16039171/lk.zip)您好，您能否为 MRD-LX2 华为 Y6 Pro 2019 修补我的 LK
> 
> lkpatcher is oppo/realme tool,i not can find base address patch this lk.img

That is not true. lkpatcher can work with any device. Please don't spread misinformation.

lopestom, 2025-06-18:
> [lk.zip](https://github.com/user-attachments/files/16039171/lk.zip) hello can u patch my lk for mrd-lx2 huawei y6 pro 2019

```json
Orange State.... 
Your device has been unlocked and can't be trusted..Your device will boot in 5 seconds
```

Try new lk.img file with device bootloader unlocked!
`fastboot flash lk lk_mod.img` 

[lk_mod.zip](https://github.com/user-attachments/files/20786269/lk_mod.zip)