R0rt1z2/lkpatcher #26: Thanks so much

SZRabinowitz, 2024-09-02:
Hi. 

This tool was extremely useful for me. I was finally able to get rid of that verity error. If there is a way for me to contribute a small donation, I would be more than happy to do so.

PS: The website was harder for me to understand. The old PDF instructions were much easier for me. (maybe because it had the specific example of verity, but I think the pictures helped a ton).

R0rt1z2, 2024-09-02:
Thank you for your kind words! I'm sorry to hear that the website wasn't clear enough. I plan to update the entire post at some point.

I don't usually receive many donations, but if you'd like, you can buy me a coffee through PayPal: paypal.me/rorle :)

SZRabinowitz, 2024-09-02:
Do not worry about it. The PDF was extremely helpful, and using the example I was able to patch my LK to remove orange state warning as well as dm verity warning. 

I likely would have never figured it out without this.

R0rt1z2, 2024-09-02:
> Do not worry about it. The PDF was extremely helpful, and using the example I was able to patch my LK to remove orange state warning as well as dm verity warning.
> 
> I likely would have never figured it out without this.

I'm glad to hear that you've managed to figure it out. If you need any help or feel like asking any specific questions, let me know. I have received the donation, thank you very much!

I'm going to close this topic, if you want to report something feel free to reopen it or create a new one.

closed 2024-09-02 by R0rt1z2