R0rt1z2/lkpatcher #28: trouble patching lk.img

smileyraver, 2024-10-27:
[lk.img.zip](https://github.com/user-attachments/files/17535050/lk.img.zip)

could you please patch my lk.img - i'm having trouble with it. 

i have only renamed it - not actually zipped it

R0rt1z2, 2024-10-28:
It seems to be working fine on my end. What specific issue are you encountering?

```bash
r0rt1z2@r0rt1z2: $ python3 -m lkpatcher lk.img
[root] [INFO] [2024-10-28 19:07:53] MediaTek bootloader (LK) patcher - version: 3.0.0 by R0rt1z2.
[lkpatcher.patcher] [INFO] [2024-10-28 19:07:53] Successfully loaded 5 patches.
[lkpatcher.patcher] [WARNING] [2024-10-28 19:07:53] Skipping invalid needle 2de9f04fadf5ac5d.
[lkpatcher.patcher] [INFO] [2024-10-28 19:07:53] Successfully applied 4 patches.
[root] [INFO] [2024-10-28 19:07:53] Patched image saved to lk-patched.img.
r0rt1z2@r0rt1z2: $ python3 -m lkpatcher lk.img
```

Just like you've did, I only renamed it rather than actually zipping it: [lk-patched.img.zip](https://github.com/user-attachments/files/17547233/lk-patched.img.zip)

smileyraver, 2024-10-29:
i will re-run it as i was on windows 11 -  it gave me invalid needle error - it didnt get to the applied patches stage  from memory and i didnt even think to paste the outcome in my issue that i posted.

i have just setup a new linux system as been meaning to for a while and been having a heap of issues with win 11 and python scripts for some reason.

i will finish setting up the linux system and try it again - i will be curious to see if if gives me the same error - but i will run it again when i boot back into win 11 and paste the outcome

but thanks

smileyraver, 2024-11-13:
Hey mate. Been so flat out with RL , haven't had a chance to run this or check if it works. 

Will close this and when I re-run it and flash th lk image I will let you know how it goes..

closed 2024-11-13 by smileyraver