R0rt1z2/lkpatcher #31: Patch LK for mt6761 tiq m5 - Red State

TripleU613, 2024-12-17:
Can't Boot After Unlocking Bootloader.
Heres the orginal signed lk  file
[lk.zip](https://github.com/user-attachments/files/18163665/lk.zip)

R0rt1z2, 2024-12-17:
Please dump and upload `expdb` so we can check what's going on.

xerootg, 2025-01-20:
Ok, i got it. used mtkclient to back everything up, flashed a magisk patched boot, empty vbemta, vbmeta_product, vbmeta_vendor, lk patched with your online patcher, then got a vbmeta corruption message, and since the power button detect in lk is not set wrong, it just powers off. so i found the video_clear_screen and just returned instead, as it would have had the power key been detected correctly.

[image]

closed 2025-02-04 by R0rt1z2