R0rt1z2/lkpatcher #32: Smartphone not turning on after patched lk flash, black screen

tommynok, 2025-02-03:
Device Tecno Camon 19Neo
CH6i
Stock 13A
After flashing the patched (online patch) lk, the device does not turn on.
Black screen, no signs of life.
Fastboot mode also doesn't work.
I managed to flash the stock lk using MTK client.
Is there a chance to overcome this and remove the orange state message?

tommynok, 2025-02-03:
> unlocked seccfg beforehand?

I hope I understood you correctly, and the question concerns unlocking the bootloader.
I unlocked the bootloader the normal way.
via fastboot

fastboot flashing unlock 

I use mtk client as an emergency utility

tommynok, 2025-02-04:
> * EXPDB

after flashing the patched LC I tried to turn on the device.
It didn't work as expected, there wasn't even any vibration.

[expdb_after_erasing.zip](https://github.com/user-attachments/files/18659847/expdb_after_erasing.zip)

R0rt1z2, 2025-02-09:
> > * EXPDB
> 
> after flashing the patched LK I tried to turn on the device. It didn't work as expected, there wasn't even any vibration.
> 
> [expdb_after_erasing.zip](https://github.com/user-attachments/files/18659847/expdb_after_erasing.zip)

```
[SEC_POLICY] reached the end, use default policy
[SBC] cert verify, part = lk_b, img = lk...ok
[PART] part: lk_b img: lk cert vfy(17 ms)
[PART] load speed: 10254KB/s, 1081700 bytes, 103ms
[SBC] img auth fail(0x101004)
fail(0x101004)
<ASSERT> /jenkins/buildsrv-ci/workspace/BAK_PY_UNIFIED_VERSION_BUILD_DOCKER/gerrit.transsion.com/code/vendor/mediatek/proprietary/bootable/bootloader/preloader/platform/mt6768/src/core/partition.c:line 208 0
PL fatal error...
PL delay for Long Press Reboot
```

Unfortunately your device uses SBC and therefore patched LKs won't work.

closed 2025-02-09 by R0rt1z2

hicode002, 2025-03-17:
> ```
> [SEC_POLICY] reached the end, use default policy
> [SBC] cert verify, part = lk_b, img = lk...ok
> [PART] part: lk_b img: lk cert vfy(17 ms)
> [PART] load speed: 10254KB/s, 1081700 bytes, 103ms
> [SBC] [...]

what is SBC?Can you tell me some information about it?