R0rt1z2/lkpatcher #40: Help with MT8189 LK Patch (have found related function)

hadesome1, 2025-08-16:
Device is MT8189，in Ghidra using "AARCH64 v8A 64 Little Default" to analyze it.

"red_state" and "orange_state" related FUN_0004d7c0
[image]

"dm-verity corruption" related FUN_0004ca14
[image]

the lk.img file uploaded
[lk.zip](https://github.com/user-attachments/files/21814073/lk.zip)

How can i modify this please?
@R0rt1z2 @lopestom

R0rt1z2, 2025-08-16:
Can you also upload expdb?

hadesome1, 2025-08-16:
> Can you also upload expdb?

sorry don't have expdb,device is broken now